buhuiming/BleCore #1: 大佬有设备会报这个错：在BleDeviced的Parcelable闪退

lichao3140, 2023-07-01:
aused by: java.lang.RuntimeException: Parcel android.os.Parcel@ff2c20f: Unmarshalling unknown type code 7209071 at offset 304
                                                                                                    	at android.os.Parcel.readValue(Parcel.java:2763)
                                                                                                    	at android.os.Parcel.readArrayMapInternal(Parcel.java:3053)
                                                                                                    	at android.os.BaseBundle.initializeFromParcelLocked(BaseBundle.java:288)
                                                                                                    	at android.os.BaseBundle.unparcel(BaseBundle.java:232)
                                                                                                    	at android.os.Bundle.getParcelable(Bundle.java:940)
                                                                                                    	at android.content.Intent.getParcelableExtra(Intent.java:7542)
                                                                                                    	at com.bhm.demo.ui.DetailOperateActivity.initData(DetailOperateActivity.kt:62)
                                                                                                    	at com.bhm.demo.BaseActivity.onCreate(BaseActivity.kt:75)
                                                                                                    	at android.app.Activity.performCreate(Activity.java:7317)
                                                                                                    	at android.app.Activity.performCreate(Activity.java:7308)
                                                                                                    	at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1272)
                                                                                                    	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3128)
                                                                                                    	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3299) 
                                                                                                    	at android.app.servertransaction.LaunchActivityItem.execute(LaunchActivityItem.java:78) 
                                                                                                    	at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:114) 
                                                                                                    	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:74) 
                                                                                                    	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1988) 
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:106) 
                                                                                                    	at android.os.Looper.loop(Looper.java:242) 
                                                                                                    	at android.app.ActivityThread.main(ActivityThread.java:7240) 
                                                                                                    	at java.lang.reflect.Method.invoke(Native Method) 
                                                                                                    	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:502) 
                                                                                                    	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:906)

buhuiming, 2023-07-03:
没遇到呢，什么情况下发生？

buhuiming, 2023-07-03:
修改了BleDevice构造函数tag parcel.readValue对应的类型，请升级新版本试一试

buhuiming, 2023-07-03:
修改了BleDevice参数，请使用最新版本

closed 2023-07-17 by buhuiming