buhuiming/BleCore #2: 支持多连接嘛？

Abvatous, 2023-07-10:
您好~
我想问下这个库支持多连接嘛？我 看了下库代码好像没发现，请问这个在需求计划表之内了吗？
看到大佬这个库用上了 kotlin 以及 jetpack 套件很是惊喜！！！ 感谢作者的努力~

buhuiming, 2023-07-10:
支持多连接，BleConnectedDeviceManager就是管理每一个连接的设备

Abvatous, 2023-07-11:
麻烦问下，这个多连接效果如何尼？这希望有具体点的测试数据支持选型。

buhuiming, 2023-07-11:
建议你体验一下demo，或者写个demo集成测试一下，这样才能更好的帮助你选型。如在使用中有遇到什么问题，非常欢迎提出来大家交流

Abvatous, 2023-07-14:
你好，我有个疑惑像咨询下：
场景：app 蓝牙 notify 指令成功后， 设备立马回复数据包， 理论上 notify 的 onCharacteristicChanged 应该显示这个回复数据包结果

结果：
BleCore 没有显示预期的结果：
[image]

以下是我的代码

```
BleManager.get().notify(bleDevice, serviceUUID, notifyUUID, false) {
            onNotifyFail {
                addLogMsg(LogEntity(Level.OFF, "notify失败：${it.message}"))
                failCall.invoke()
            }
            onNotifySuccess {
                addLogMsg(LogEntity(Level.FINE, "notify成功：${notifyUUID}"))
            }
            onCharacteristicChanged {
                Log.e("bobo", "notify: " + BleUtil.bytesToHex(it))
                addLogMsg(
                    LogEntity(
                        Level.INFO, "Notify接收到${notifyUUID}的数据：" +
                                BleUtil.bytesToHex(it)
                    )
                )
            }
        }
```
<br>
<br>
原来的 Ble SDK 是能显示这个 notify 的回复数据包：

``` 
BleManager.getInstance().notify(bleDevice,
            characteristic.service?.uuid.toString(),
            characteristic.uuid.toString(),
            object : BleNotifyCallback() {
                override fun onNotifySuccess() {
                    addText(textCommandRes, versionDate.format(Date()) + ": notify 打开 success")
                }

                override fun onNotifyFailure(exception: BleException?) {
                    CommandHolder.getInstance().deInitCommand()
                }

                override fun onCharacteristicChanged(data: ByteArray?) {
                    Log.e("bobo", HexUtil.formatHexString(data, true))
                    addText(textCommandRes,  HexUtil.formatHexString(
                            data, true))
                }
            })
```

[image]

其中 4b 45 59 4b f9 eb 34 和 48 41 4c 3d 31 就是蓝牙模块 notify 成功后立马发送给 app 的内容，但是 BleCore 的 onCharacteristicChanged 没有是出现，请问是需要配置参数嘛？

buhuiming, 2023-07-14:
我未遇到这个问题，方便加我qq[873247376]联系嘛？

buhuiming, 2023-07-14:
已解决，请更新一下，v1.0.2

closed 2023-07-17 by buhuiming