buhuiming/BleCore #11: 关于短时间内写入大量数据ANR的问题

JOJOyh, 2023-09-18:
```
ANR in com.bhm.ble (com.bhm.ble/com.bhm.demo.ui.DetailOperateActivity)
                                                                                                    PID: 32329
                                                                                                    Reason: Input dispatching timed out (5b0d78d com.bhm.ble/com.bhm.demo.ui.DetailOperateActivity (server) is not responding. Waited 5002ms for MotionEvent)
                                                                                                    Parent: com.bhm.ble/com.bhm.demo.ui.DetailOperateActivity
                                                                                                    Frozen: false
                                                                                                    Load: 0.0 / 0.0 / 0.0
                                                                                                    ----- Output from /proc/pressure/memory -----
                                                                                                    some avg10=0.00 avg60=0.00 avg300=0.00 total=3050098580
                                                                                                    full avg10=0.00 avg60=0.00 avg300=0.00 total=1236302433
                                                                                                    ----- End output from /proc/pressure/memory -----
                                                                                                    
                                                                                                    CPU usage from 0ms to 6611ms later (2023-09-18 18:18:29.225 to 2023-09-18 18:18:35.836):
                                                                                                      30% 1382/system_server: 13% user + 16% kernel / faults: 10811 minor
                                                                                                      4.2% 1550/media.codec: 2.9% user + 1.3% kernel / faults: 32066 minor
                                                                                                      5.9% 1128/surfaceflinger: 3.6% user + 2.2% kernel / faults: 988 minor
                                                                                                      5.1% 17486/adbd: 0.6% user + 4.5% kernel / faults: 10 minor
                                                                                                      4.5% 11643/com.jhyan.yan:widgetProvider: 1% user + 3.4% kernel / faults: 142 minor
                                                                                                      4% 1037/android.hardware.sensors@2.0-service.multihal: 2.2% user + 1.8% kernel / faults: 352 minor
                                                                                                      3.3% 669/kworker/u24:7-memlat_wq: 0% user + 3.3% kernel
                                                                                                      3.1% 3539/com.android.phone: 1.9% user + 1.2% kernel / faults: 1223 minor
                                                                                                      2.7% 626/logd: 0.3% user + 2.4% kernel
                                                                                                      2.2% 668/kworker/u24:6-events_unbound: 0% user + 2.2% kernel
                                                                                                      0.3% 1490/media.extractor: 0.1% user + 0.1% kernel / faults: 3681 minor
                                                                                                      2.2% 12235/com.sohu.inputmethod.sogouoem: 1.6% user + 0.6% kernel / faults: 1162 minor
                                                                                                      2.2% 32213/com.android.systemui: 1.9% user + 0.3% kernel / faults: 110 minor
                                                                                                      2.1% 664/kworker/u24:1+memlat_wq: 0% user + 2.1% kernel
                                                                                                      2.1% 24292/kworker/u24:2-events_unbound: 0% user + 2.1% kernel
                                                                                                      2.1% 26537/kworker/u25:3-fsverity_read_queue: 0% user + 2.1% kernel
                                                                                                      0% 1014/android.hardware.camera.provider@2.4-service_64: 0% user + 0% kernel / faults: 30 minor
                                                                                                      1.8% 32329/com.bhm.ble: 1.2% user + 0.6% kernel / faults: 1727 minor
                                                                                                      0.2% 25222/com.android.bluetooth: 0% user + 0.2% kernel / faults: 261 minor
                                                                                                      1% 17645/process-tracker: 0.1% user + 0.9% kernel
                                                                                                      0.9% 11/rcu_preempt: 0% user + 0.9% kernel
                                                                                                      0.9% 470/usbtemp_kthread: 0% user + 0.9% kernel
                                                                                                      0% 5572/com.qualcomm.qti.workloadclassifier: 0% user + 0% kernel / faults: 395 minor
                                                                                                      0.9% 21604/process-tracker: 0% user + 0.9% kernel / faults: 2 minor
                                                                                                      0% 722/tombstoned: 0% user + 0% kernel
                                                                                                      0.7% 4896/irq/52-90b6300.: 0% user + 0.7% kernel
                                                                                                      0.7% 17711/com.quark.browser:channel: 0.6% user + 0.1% kernel / faults: 391 minor
                                                                                                      0.7% 24889/kworker/u25:1-dwc_wq: 0% user + 0.7% kernel
                                                                                                      0.6% 293/kgsl_worker_thr: 0% user + 0.6% kernel
                                                                                                      0.6% 983/qrtr_rx: 0% user + 0.6% kernel
                                                                                                      0.6% 1050/vendor.qti.hardware.display.composer-service: 0% user + 0.6% kernel / faults: 251 minor
                                                                                                      0.6% 17638/logcat: 0.1% user + 0.4% kernel
                                                                                                      0.6% 19844/kworker/u24:3-events_unbound: 0% user + 0.6% kernel
                                                                                                      0.6% 21601/logcat: 0.1% user + 0.4% kernel
                                                                                                      0.6% 31866/logcat: 0.1% user + 0.4% kernel
                                                                                                      0.4% 14/rcuop/0: 0% user + 0.4% kernel
                                                                                                      0.4% 37/rcuop/2: 0% user + 0.4% kernel
                                                                                                      0% 862/zygote64: 0% user + 0% kernel / faults: 22 minor
                                                                                                      0.4% 2084/com.taobao.taobao:channel: 0.3% user + 0.1% kernel / faults: 1 minor
                                                                                                      0.4% 4584/com.tencent.mm: 0.1% user + 0.3% kernel / faults: 19 minor
                                                                                                      0.4% 4898/irq/53-90cd000.: 0% user + 0.4% kernel
[... cut ...]
            }
            currentSendNode?.let { node ->
                lifecycleScope.launch {
                    withContext(Dispatchers.IO){
                        repeat(1000) {
                            viewModel.writeData(
                                getBleDevice(),
                                node,
                                content
                            )
                        }
                    }
                }
            }
        }
```

如果复现不出来，可以尝试点击界面中的其他按钮

JOJOyh, 2023-09-18:
而且如果加入延迟100ms，用手指移动去产生数据，发送蓝牙命令，超过一定时间后，就不会产生数据了，不知道为啥，都不会调用writedata这个方法了

buhuiming, 2023-09-19:
1、在执行写操作的时候，应该避免其他的操作，比如点击其他按钮notify、设置mtu等
2、如题所遇到的ANR，原因是在执行写操作的时候，执行了其他的操作，导致蓝牙资源抢占而忙碌所引起的超时。(你可以试一下点击“清除日志”按钮，或者写一个弹toast按钮，或者滑动列表，是不会导致ANR的)。
3、demo中示例仍存在缺陷，具体业务需求应充分考虑应用场景

closed 2023-10-07 by buhuiming