buhuiming/BleCore #14: 关于接收大量数据的问题

vistar1986, 2023-11-01:
作者你好，在使用过程中接收大量数据时会出现数据错乱的问题

[image]

在接收大数据时，onCharacteristicChanged接收的某包数据会延时发出，导致后续组帧错误

vistar1986, 2023-11-01:
[image]
这是在onCharacteristicChanged中打印日志的地方，打印日志前没有做任何耗时操作

buhuiming, 2023-11-02:
麻烦更新到v1.5.0，把日志去掉再试试看

vistar1986, 2023-11-02:
> 麻烦更新到v1.5.0，把日志去掉再试试看

已经升级到v1.5.0，并且把日志关闭，还是会出现问题~

[image]
[image]

以下为导入代码
[image]
[image]

buhuiming, 2023-11-02:
我大概知道什么原因了，onCharacteristicChanged被指定在IO线程，子线程不保证顺序。我已经去掉IO线程的指定，如果有需要在IO线程处理，业务端自行处理。
请更新1.5.1

vistar1986, 2023-11-02:
> 我大概知道什么原因了，onCharacteristicChanged被指定在IO线程，子线程不保证顺序。我已经去掉IO线程的指定，如果有需要在IO线程处理，业务端自行处理。 请更新1.5.1

你好，在更新新版后，数据暂时没有出现错乱的情况，我会持续测试，感谢作者大大

buhuiming, 2023-11-02:
好的，感谢你的反馈

closed 2023-11-06 by buhuiming