buhuiming/BleCore #17: 数据频繁交互时出现卡死

vistar1986, 2023-11-27:
作者你好：
  在使用过程中，实现一秒查询一次数据，返回数据量2000个字节左右，如果长时间进行数据交互会偶发性出现蓝牙Notifiy无法收到数据的情况，但是发送数据时没有问题的，当这个时候断开蓝牙连接会出现ANR报错。

buhuiming, 2023-11-28:
在问题排查之前，请确保更新最新版本，确保onCharacteristicChanged方法在处理数据时所属线程，请提供版本号、运行环境、代码示例，否则无法更好的定位。

wuxinxi, 2023-12-17:
> 作者你好： 在使用过程中，实现一秒查询一次数据，返回数据量2000个字节左右，如果长时间进行数据交互会偶发性出现蓝牙Notifiy无法收到数据的情况，但是发送数据时没有问题的，当这个时候断开蓝牙连接会出现ANR报错。

你那边解决了吗？onCharacteristicChanged的确是在子线程，当长时间频繁读写数据时最新代码也会出现

> 在问题排查之前，请确保更新最新版本，确保onCharacteristicChanged方法在处理数据时所属线程，请提供版本号、运行环境、代码示例，否则无法更好的定位。

- 最新版本
- onCharacteristicChanged：子线程
- 长时间频繁读写数据时，Notifiy会偶尔收不到消息，此时再去断开蓝牙就会出现ANR
- 貌似是死锁了

wuxinxi, 2023-12-17:
> > 作者你好： 在使用过程中，实现一秒查询一次数据，返回数据量2000个字节左右，如果长时间进行数据交互会偶发性出现蓝牙Notifiy无法收到数据的情况，但是发送数据时没有问题的，当这个时候断开蓝牙连接会出现ANR报错。
> 
> 你那边解决了吗？onCharacteristicChanged的确是在子线程，当长时间频繁读写数据时最新代码也会出现
> 
> > 在问题排查之前，请确保更新最新版本，确保onCharacteristicChanged方法在处理数据时所属线程，请提供版本号、运行环境、代码示例，否则无法更好的定位。
> 
> * 最新版本
> * onCharacteristicChanged：子线程
> * 长时间频繁读写数据时，Notifiy会偶尔收不到消息，此时再去断开蓝牙就会出现ANR
> * 会有写入数据超时的日志

closed 2023-12-18 by buhuiming

vistar1986, 2023-12-19:
版本：1.6.2
手机：华为NOVA6
复现：页面中有一个开关按钮，点击一次会发送一次蓝牙写入请求，当页面不做任何限制时，频繁点击该按钮发送数据会发生卡死现象
以下为蓝牙发送代码片段：
[image]
以下为页面截图：
[image]

vistar1986, 2023-12-19:
以下为卡死时的日志记录
[image]

wuxinxi, 2023-12-19:
频繁的数据交互还是用bluetoothSocket吧

buhuiming, 2024-03-06:
初步诊断为Synchronized和Collections.synchronizedList资源竞争死锁问题导致ANR，已优化，请更新最新版本