buhuiming/BleCore #32: 为什么大于等于Android S还需要位置权限

zdxiang, 2024-04-07:
你的模板不适合我

buhuiming, 2024-04-08:
Android S蓝牙权限适配我没有做得很详细，如果不需要位置信息，可以不用位置权限，如果需要请Fork修改

zdxiang, 2024-04-08:
> Android S蓝牙权限适配我没有做得很详细，如果不需要位置信息，可以不用位置权限，如果需要请Fork修改

好的，我已自行解决。建议把权限判断这块抛给使用者。
 `    fun startScan(
        scanMillisTimeOut: Long?,
        scanRetryCount: Int?,
        scanRetryInterval: Long?,
        bleScanCallback: BleScanCallback
    ) {
       // if (!BleUtil.isPermission(getBleManager().getContext())) {
       //     bleScanCallback.callScanFail(BleScanFailType.NoBlePermission)
       //     return
       //}
        initScannerAndStart(
            scanMillisTimeOut,
            scanRetryCount,
            scanRetryInterval,
            bleScanCallback
        )
    }`

buhuiming, 2024-04-10:
好的

closed 2024-04-22 by buhuiming