buhuiming/BleCore #33: 特征值notify没收到onCharacteristicChanged回调（同样操作fastble可以）

chengxinping, 2024-04-10:
问题描述(出现的环境，demo是否存在)
 公司的设备两个特征值，一个用来写，一个用来notiffy，本框架写了之后notify的那个特征值收不到change的回调，fastble可以

[image]
[image]
[image]

框架文档是否提及了该问题【必答】 否

是否已经查阅框架文档但还未能解决的【必答】 **是**

issue 列表中是否有人曾提过类似的问题【必答】 **否**

是否已经搜索过了 issue 列表但还未能解决的【必答】 **是**

buhuiming, 2024-04-11:
小伙子我没有复现你的bug啊
[image]
[image]
[image]

chengxinping, 2024-04-12:
我这边只有我这款型号的设备能复现，其他设备也没法复现，大概率和机器的蓝牙芯片有关系，没办法我只能替换回fastble了

chengxinping, 2024-04-12:
我们这边这款型号的机器是必现的 脑壳痛

buhuiming, 2024-04-12:
和fastble是一样的原生Api，只不过android sdk版本不一样

buhuiming, 2024-04-23:
[image]
 小伙子，把137行、157~159号注释掉试一试

chengxinping, 2024-04-26:
尝试了一下没用，还是收不到notify的问题，咱们这个库要传递一个DescriptorGetType，fastble的notify没这个参数，不太懂这个参数是干啥的，不知道是不是这个参数影响的 我看你这个参数里面注释也说了这个问题

`开发过程中，我们会遇到不同硬件设备定义UUID的情况，有的硬件设备通过特征值的UUID来获取描述符(用来writeDescriptor，
 * 打开或关闭notify、indicate)，而非是通过系统提供接受通知自带的UUID获取描述符。此外特征值有多个描述符时，获取其中
 * 一个描述符来写入数据，可能会导致onCharacteristicChanged函数没有回调，我不确定是否是硬件设备需要支持修改的问题。`

buhuiming, 2024-04-28:
1、BleDescriptorGetType有3个类型，除了AllDescriptor，其他2个是和FaseBle对应的。AllDescriptor这个其实把该特征值下的所有描述符都写入。Default是系统提供接受通知自带的UUID对应的特征值，和FaseBle useCharacteristicDescriptor=fasle一样。CharacteristicDescriptor是业务层指定的UUID对应的特征值，和FaseBle useCharacteristicDescriptor=true一样。不知道你有没有3个值都试一下？
2、第二个可能会导致，就是BleConnectedDevice类中的onCharacteristicChanged方法，Android13中引用了新的方法，如果新方法和旧方法都没有触发的话，这个问题目前就无解了

closed 2024-06-03 by buhuiming