buhuiming/BleCore #43: 小米手机接收不到数据，华为和pixel手机能接收到

kllkko, 2024-07-29:
问题描述(出现的环境，demo是否存在)【必填】

我们有个蓝牙接口，需要先发送1个数据，表示偏移量，然后读取数据（这个数据量比较大，后台准备数据会花几秒钟时间），看log小米手机会直接读出空，华为和pixel手机能正常读取到，demo也存在

框架文档是否提及了该问题【必答】
否

是否已经查阅框架文档但还未能解决的【必答】
是

issue 列表中是否有人曾提过类似的问题【必答】
否

是否已经搜索过了 issue 列表但还未能解决的【必答】
是

下面是小米手机log
```
2024-07-29 18:18:38.648 21365-21365 DetailViewModel         com.bhm.ble                          I  #######----> data is: 31 
2024-07-29 18:18:38.652 21365-21365 BleTaskQueue            com.bhm.ble                          D  #######----> (共享队列) 当前任务数量：0, 添加任务：com.bhm.ble.control.BleTask@3bf851b
2024-07-29 18:18:38.758 21365-21440 BleTaskQueue            com.bhm.ble                          I  #######----> (共享队列) 开始执行任务：com.bhm.ble.control.BleTask@3bf851b
2024-07-29 18:18:38.759 21365-21440 BleWriteRequest         com.bhm.ble                          I  #######----> 1005：00003a00-0000-1000-8000-00805f9b34fb(1722248318649)(1) - > 开始写第1包数据
2024-07-29 18:18:38.766 21365-21386 BleWriteRequest         com.bhm.ble                          I  #######----> 1005：00003a00-0000-1000-8000-00805f9b34fb(1722248318649)(1) -> 第1包数据写成功：31 
2024-07-29 18:18:38.767 21365-21386 BleTaskQueue            com.bhm.ble                          E  #######----> (共享队列) 移除正在执行的任务：com.bhm.ble.control.BleTask@3bf851b
2024-07-29 18:18:38.768 21365-21440 BleLogger               com.bhm.ble                          I  #######----> (共享队列) 任务完成，未超时：com.bhm.ble.control.BleTask@3bf851b
2024-07-29 18:18:38.769 21365-21437 BleTaskQueue            com.bhm.ble                          D  #######----> (共享队列) 任务：com.bhm.ble.control.BleTask@3bf851b结束完毕，剩下0个任务
2024-07-29 18:18:38.770 21365-21437 BleTaskQueue            com.bhm.ble                          I  #######----> (共享队列) 所有任务执行完毕
2024-07-29 18:18:38.811 21365-21365 ScrollerOp...ionManager com.bhm.ble                          D  registerConfigChangedListener
2024-07-29 18:18:46.625 21365-21365 BleTaskQueue            com.bhm.ble                          D  #######----> (共享队列) 当前任务数量：0, 添加任务：com.bhm.ble.control.BleTask@4d79dfc
2024-07-29 18:18:46.730 21365-21437 BleTaskQueue            com.bhm.ble                          I  #######----> (共享队列) 开始执行任务：com.bhm.ble.control.BleTask@4d79dfc
2024-07-29 18:18:49.823 21365-21386 BleTaskQueue            com.bhm.ble                          E  #######----> (共享队列) 移除正在执行的任务：com.bhm.ble.control.BleTask@4d79dfc
2024-07-29 18:18:49.824 21365-21437 BleLogger               com.bhm.ble                          I  #######----> (共享队列) 任务完成，未超时：com.bhm.ble.control.BleTask@4d79dfc
2024-07-29 18:18:49.825 21365-21440 BleTaskQueue            com.bhm.ble                          D  #######----> (共享队列) 任务：com.bhm.ble.control.BleTask@4d79dfc结束完毕，剩下0个任务
2024-07-29 18:18:49.825 21365-21386 BleReadRequest          com.bhm.ble                          D  #######----> 00003a00-0000-1000-8000-00805f9b34fb -> 读特征值数据成功：
2024-07-29 18:18:49.826 21365-21440 BleTaskQueue            com.bhm.ble                          I  #######----> (共享队列) 所有任务执行完毕
2024-07-29 18:18:49.860 21365-21365 ScrollerOp...ionManager com.bhm.ble                          D  registerConfigChangedListener
```

下面是pixel手机log：
```
2024-07-29 18:22:49.548 24833-24833 DetailViewModel         com.bhm.ble                          I  #######----> data is: 31 
2024-07-29 18:22:49.550 24833-24833 BleTaskQueue            com.bhm.ble                          D  #######----> (共享队列) 当前任务数量：0, 添加任务：com.bhm.ble.control.BleTask@6d3796
2024-07-29 18:22:49.655 24833-24895 BleTaskQueue            com.bhm.ble                          I  #######----> (共享队列) 开始执行任务：com.bhm.ble.control.BleTask@6d3796
2024-07-29 18:22:49.657 24833-24895 BleWriteRequest         com.bhm.ble                          I  #######----> 1005：00003a00-0000-1000-8000-00805f9b34fb(1722248569548)(1) - > 开始写第1包数据
2024-07-29 18:22:49.668 24833-24911 BleWriteRequest         com.bhm.ble                          I  #######----> 1005：00003a00-0000-1000-8000-00805f9b34fb(1722248569548)(1) -> 第1包数据写成功：31 
2024-07-29 18:22:49.673 24833-24911 BleTaskQueue            com.bhm.ble                          E  #######----> (共享队列) 移除正在执行的任务：com.bhm.ble.control.BleTask@6d3796
2024-07-29 18:22:49.673 24833-24895 BleLogger               com.bhm.ble                          I  #######----> (共享队列) 任务完成，未超时：com.bhm.ble.control.BleTask@6d3796
2024-07-29 18:22:49.675 24833-24895 BleTaskQueue            com.bhm.ble                          D  #######----> (共享队列) 任务：com.bhm.ble.control.BleTask@6d3796结束完毕，剩下0个任务
2024-07-29 18:22:49.676 24833-24895 BleTaskQueue            com.bhm.ble                          I  #######----> (共享队列) 所有任务执行完毕
2024-07-29 18:22:51.648 24833-24833 BleTaskQueue            com.bhm.ble                          D  #######----> (共享队列) 当前任务数量：0, 添加任务：com.bhm.ble.control.BleTask@6a905b3
2024-07-29 18:22:51.754 24833-24895 BleTaskQueue            com.bhm.ble                          I  #######----> (共享队列) 开始执行任务：com.bhm.ble.control.BleTask@6a905b3
2024-07-29 18:22:54.842 24833-24848 BleTaskQueue            com.bhm.ble                          E  #######----> (共享队列) 移除正在执行的任务：com.bhm.ble.control.BleTask@6a905b3
2024-07-29 18:22:54.844 24833-24895 BleLogger               com.bhm.ble                          I  #######----> (共享队列) 任务完成，未超时：com.bhm.ble.control.BleTask@6a905b3
2024-07-29 18:22:54.844 24833-24907 BleTaskQueue            com.bhm.ble                          D  #######----> (共享队列) 任务：com.bhm.ble.control.BleTask@6a905b3结束完毕，剩下0个任务
2024-07-29 18:22:54.844 24833-24907 BleTaskQueue            com.bhm.ble                          I  #######----> (共享队列) 所有任务执行完毕
2024-07-29 18:22:54.858 24833-24848 BleReadRequest          com.bhm.ble                          D  #######----> 00003a00-0000-1000-8000-00805f9b34fb -> 读特征值数据成功：7b 22 73 73 69 64 22 3a 22 4c 44 43 2d 6d 61 6e 61 67 65 72 2d 61 70 22 2c 22 6b 65 79 22 3a 22 22 2c 22 65 6e 63 72 79 70 74 69 6f 6e 22 3a 22 70 73 6b 32 22 2c 22 73 69 67 6e 61 6c 4c 65 76 65 6c 22 3a 22 2d 35 35 22 2c 22 63 6f 6e 6e 65 63 74 65 64 22 3a 66 61 6c 73 65 7d 2c 7b 22 73 73 69 64 22 3a 22 4c 44 43 2d 6f 66 66 69 63 65 2d 74 65 73 74 31 22 2c 22 6b 65 79 22 3a 22 22 
```

kllkko, 2024-07-29:
如果后台数据很小，不用准备很久，就会读正常

buhuiming, 2024-07-30:
mtu设置多少呢

kllkko, 2024-07-30:
> mtu设置多少呢

作者大大你好，是直接调用的BleManager.get().setMtu方法，没有指定具体的数值，我看小米log是`设置Mtu成功：23`  ，pixel log是`设置Mtu成功：256`

kllkko, 2024-07-30:
小米我设置全局mtu为256，会提示   设置mtu值失败：A8:96:09:BB:50:5B -> 设置Mtu超时

buhuiming, 2024-07-30:
如果没有调用设置mtu，系统有默认的。默认值是根据手机系统决定的

kllkko, 2024-07-31:
> 如果没有调用设置mtu，系统有默认的。默认值是根据手机系统决定的

如果不调用设置mtu，我发现也会读不到数据，很奇怪

closed 2024-08-13 by buhuiming