buhuiming/BleCore #44: oppo reno11 setScanServiceUuid后搜索不到指定的设备，去掉setScanServiceUuid后就可以搜到，其他手机正常

kllkko, 2024-08-13:
问题描述(出现的环境，demo是否存在)【必填】
存在
设备蓝牙是开着的，在demo代码中添加了.setScanServiceUuid("00001801-0000-1000-8000-00805f9b34fb")，其他手机都是正常能搜索到这个设备，但是目前发现oppo reno11搜索不到设备，如果我把setScanServiceUuid去掉，就又可以搜索到

框架文档是否提及了该问题【必答】
否

是否已经查阅框架文档但还未能解决的【必答】
是

issue 列表中是否有人曾提过类似的问题【必答】
否

是否已经搜索过了 issue 列表但还未能解决的【必答】
是

kllkko, 2024-08-13:
```
#######----> ble Successful initialization
#######----> 注册系统蓝牙广播
#######----> 设备是否支持蓝牙: true
#######----> 获取到了权限
#######----> 开始第1次扫描
#######----> onScanStart
#######----> 没有扫描到数据
#######----> 扫描完毕，扫描次数1次
#######----> 设备是否支持蓝牙: true
#######----> 获取到了权限
#######----> 开始第1次扫描
#######----> onScanStart
#######----> 没有扫描到数据
#######----> 扫描完毕，扫描次数1次
```

kllkko, 2024-08-13:
下面是去掉setScanServiceUuid，搜索到我们设备的信息：
BleDevice(deviceInfo=XX:XX:XX:XX:50:5B, deviceName=FX_F1_1604, deviceAddress=A8:96:09:BB:50:5B, rssi=-54, timestampNanos=87126771292421, scanRecord=[2, 1, 6, 17, 7, -5, 52, -101, 95, -128, 0, 0, -128, 0, 16, 0, 0, 1, 24, 0, 0, 11, 9, 70, 88, 95, 70, 49, 95, 49, 54, 48, 52, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], tag=null, serviceUuids=[00001801-0000-1000-8000-00805f9b34fb])

kllkko, 2024-08-13:
作者大大，看还需要我提供什么信息，手机在我手边，随时可以提供

buhuiming, 2024-08-13:
其他调试工具有遇到这个情况吗

kllkko, 2024-08-20:
> 其他调试工具有遇到这个情况吗

作者大大有推荐的工具吗

buhuiming, 2024-08-20:
LightBlue

kllkko, 2024-08-27:
其他工具我没发现有设置setScanServiceUuid的地方，这个问题我这边暂时以.setScanDeviceName("fx_","FX_")方式过滤得以解决

kllkko, 2024-08-29:
> 还有个办法是，不使用setScanServiceUuid，在扫描回调的接口onLeScan中，对bleDevice的serviceUuids属性进行过滤

好的，谢谢作者大大

closed 2024-09-09 by buhuiming