buhuiming/BleCore #46: writeQueueData 函数的 current 一直是1

Yuanarcheannovice, 2024-09-08:
两个问题：
writeQueueData 函数的 current 一直是1;(writeData 函数是正常的;)

writeQueueData 比writeData 写入 慢，

我这边测试，包size为500的时候，writeQueueData 比writeData 写入 慢 120毫秒

问题描述(出现的环境，demo是否存在)【必填】
demo中没有使用这个函数 writeQueueData

框架文档是否提及了该问题【必答】
没有

是否已经查阅框架文档但还未能解决的【必答】
是的

issue 列表中是否有人曾提过类似的问题【必答】
没有

是否已经搜索过了 issue 列表但还未能解决的【必答】
是的

buhuiming, 2024-09-09:
writeQueueData是在writeData的基础上添加了一个队列，所以writeData接受到的数据永远是只有一个，因此current和total都会是1，这里是不好处理的，因为writeQueueData的队列是支持动态添加的(也就是队列中还有未写的数据，这时还能添加追加数据在队列中)。

Yuanarcheannovice, 2024-09-09:
那我自己封装一层，作为计数；
但是 写入速度，也比writeData慢；
写完后，没有回调；

buhuiming, 2024-09-09:
如果使用writeQueueData，需要业务端自行处理写进度(总进度)。 第二个writeQueueData正常来说比writeData要快，除非writeData设置的写间隔非常小(弊端就是设备写忙碌)。writeQueueData是第一个包写成功之后，才会写第二个包，而writeData只是设置了每个包之间的间隔，第一个包是否成功，第二个包都会在间隔时间后写

buhuiming, 2024-09-09:
建议OTA或者要求写入成功率，使用writeQueueData。其他情况可以使用writeData

Yuanarcheannovice, 2024-09-09:
明白了；
writeData 写入间隔是默认的100毫秒；我这边的写入速度是 170毫秒左右；
writeQueueData，因为需要等待是否写入成功，所以慢了120毫秒，大概300毫秒才能写入一次；
每次写入500字节为例，500k的ota固件包，要写入十多分钟，硬件那边觉得写入太慢了;

buhuiming, 2024-09-09:
你们固件写入flash是不是太慢了，当然这个跟硬件条件有关。另外还有个条件是跟手机的BLE性能有关。我这里有个产品，1M的文件，ota使用的是writeQueueData方法，大多手机大概45秒，有的手机要75秒

buhuiming, 2024-09-09:
writeData 写入间隔是默认的100毫秒，这个是可以自行设置的

buhuiming, 2024-09-09:
OTA的时候，writeData 写入间隔的间隔最好多调试一下，如果直接使用writeQueueData，写间隔可以设置为5ms

Yuanarcheannovice, 2024-09-09:
writeData 写入间隔是默认的100毫秒，这个是可以自行设置的, 这个知道，已设置，也设置了高性能模式；
writeQueueData的话，我设置的500个字节写入一次，没问题吧？
如果这个没有问题的话，那剩下的速度问题，就是硬件那边了；

buhuiming, 2024-09-09:
只要设置了MTU，500个字节写入一次没问题。同时把写间隔写成5ms~10ms，也就是operateInterval参数

buhuiming, 2024-09-09:
可以下载杰里的apk来调试固件OTA。 我这里一直用这个来做对比，基本上都是比杰里的要快

Yuanarcheannovice, 2024-09-09:
感谢解答

buhuiming, 2024-09-09:
我这边OTA使用的是writeQueueData。其他写数据都是用writeData，使用writeQueueData之前，设置operateInterval=5ms，使用完之后，恢复operateInterval=100ms

Yuanarcheannovice, 2024-09-09:
writeQueueData 写完的回调，也是业务端处理的吗？自己计算次数，然后判断是否完成写入？

buhuiming, 2024-09-14:
是的，writeQueueData无法处理，因为writeQueueData的队列是支持可变，current和total是不准确的

closed 2024-09-14 by Yuanarcheannovice