buhuiming/BleCore #48: 概率性出现任务队列没有执行就移除

zpliang94, 2024-09-21:
问题描述(出现的环境，demo是否存在)
demo没有出现，一次性进行了10来次写任务，有小概率出现

框架文档是否提及了该问题【必答】
没有
是否已经查阅框架文档但还未能解决的【必答】
是的
issue 列表中是否有人曾提过类似的问题【必答】
没有
是否已经搜索过了 issue 列表但还未能解决的【必答】
未能解决

图片中第4个任务BleTask@eacb996 没有进行写数据就被移除了，这问题小概率出现
[image]
[image]

buhuiming, 2024-10-09:
你是通过什么代码来触发发送数据的，看日志是因为有2个数据包，通过getTaskId来生成了两个一样的taskid导致的

zpliang94, 2024-10-11:
使用 BleManager.get().writeData()  发送数据，调用了十多个数据包 。
现在使用 BleManager.get().writeQueueData(mBleDevice, KeyUtils.UUID_SERVICE_E, KeyUtils.UUID_WRITE, byteArray, retryWriteCount = 1 ) 方法就没出现过这情况。 @buhuiming

zpliang94, 2024-10-11:
for循环调用 BleManager.get().writeData() 发送数据

buhuiming, 2024-10-12:
好的，不建议for循环调用BleManager.get().writeData() ，这个方法是支持传递数据集合的

zpliang94, 2024-10-16:
连续调用 BleManager.get().writeQueueData()，BleManager.get().writeQueueData()，BleManager.get().writeQueueData()  还是会出现taskId 一样，有些指令没有发送就被移除

zpliang94, 2024-10-16:
主要原因是生成的时间戳是一样

zpliang94, 2024-10-18:
@buhuiming id还是一样的
[image]

[image]

[image]

zpliang94, 2024-10-18:
更新了v2.3.1版本，连续调用 BleManager.get().writeQueueData() 三次，BleConnectedDevice.writeQueueData() 方法中id不一样，bleWriteRequest?.writeQueueData() 的taskId只以第一次为id @buhuiming

zpliang94, 2024-10-18:
第二次使用writeQueueData()的id还是为第一次writeQueueData()的id 1859999209893252jRY8

buhuiming, 2024-10-18:
你先用writeData方法，这个应该没问题。我再看看writeQueueData

buhuiming, 2024-10-18:
找到原因了，下周我调试修复再更新

zpliang94, 2024-10-20:
重看了第一次提交问题的log，好像不是taskid相同的原因

zpliang94, 2024-10-20:
还有一个可能点是 operateInterval ，设置200时候也容易出现没有执行就移除

buhuiming, 2024-10-21:
请更新到v2.3.2再试试

closed 2024-10-24 by zpliang94