buhuiming/BleCore #53: 设备主动断开蓝牙，客户端重新连接后，短时间内多次发消息

Yuanarcheannovice, 2025-03-17:
问题描述(出现的环境，demo是否存在)【必填】
否
框架文档是否提及了该问题【必答】
否
是否已经查阅框架文档但还未能解决的【必答】
是
issue 列表中是否有人曾提过类似的问题【必答】
否
是否已经搜索过了 issue 列表但还未能解决的【必答】
是

设备主动断开蓝牙，客户端重新连接后，短时间内多次发消息，硬件反馈500ms内发了6，7条消息，必现，
我推测是 read队列里面 因为read失败，重试导致的，
有没有办法，搞个 变量，判断设备断开后，不需要重连，不需要重试read，或者write，清理队列

buhuiming, 2025-03-18:
断开连接后调用Blemanager.get().close(device)

Yuanarcheannovice, 2025-03-18:
> 断开连接后调用Blemanager.get().close(device)

调用了的，没啥用

buhuiming, 2025-03-19:
我这里没有复现，你是使用writeData还是writeQueueData方法，麻烦提供一下日志

Yuanarcheannovice, 2025-03-19:
> 我这里没有复现，你是使用writeData还是writeQueueData方法，麻烦提供一下日志

[image]

[image]

读的，2秒读一次，日志暂时没办法打印，硬件那边在改设备，要过几天

closed 2025-04-14 by buhuiming