buhuiming/BleCore #56: 华为手机后台回前台问题

Yuanarcheannovice, 2025-04-10:
问题描述(出现的环境，demo是否存在)【必填】
否
框架文档是否提及了该问题【必答】
否
是否已经查阅框架文档但还未能解决的【必答】
是
issue 列表中是否有人曾提过类似的问题【必答】
否
是否已经搜索过了 issue 列表但还未能解决的【必答】
是

华为mate40pro  nova10 ，已确定这两款有问题，其他没试，
操作步骤，打开开发者选项中的 不保留activity，然后打开app，连接设备，2秒读一次设备信息，
然后去开启悬浮窗权限，或者去打电话（把app置于后台一下），
再打开app，此时页面被销毁重建，BleDevice对象我看了内存地址没变，获取已连接的蓝牙也正常，设备也没有断开蓝牙连接的表示，但是app无法跟设备进行任何交互；
我通过app进行断开连接，然后重新连接设备，就正常了

这是日志

[app_log22.txt](https://github.com/user-attachments/files/19678327/app_log22.txt)

buhuiming, 2025-04-14:
那是因为系统将BleCore的资源回收掉了。你有使用Service服务去保证BleCore的后台运行吗

Yuanarcheannovice, 2025-04-14:
Foreground Service吗？  也是有可能被回收的吧？
复现多次后，read有时候没有问题，notify必出超时
有联系方式吗？有偿请教

buhuiming, 2025-04-14:
[873247376@qq.com](mailto:873247376@qq.com)

closed 2025-04-14 by Yuanarcheannovice